luizribeiro/mariner #493: Unable to finish setup.

Viii0Viii, 2021-12-28:
I am using a Raspberry PI 4B

Power is supplied via the header

I have the USB-C port wired into the USB port of the ChituBoard

Serial connection has been verified RX to TX, TX to RX, and GND to GND

RaspiOS was installed via Raspberry Pi Imager using the Raspberry Pi OS Lite (32-Bit)

I have followed many guides, including this one, and the auto installer.

  if mention(auto installer)=https://github.com/KTheMan/Mariner-Autoinstaller

I have also followed the guide trying to update/create a symbolic link from Python 3.7 to Python 3.9

  if mention(python)=https://github.com/luizribeiro/mariner/issues/481#issuecomment-968325624

No matter, I still have ended up with no web GUI and the status showing as below.

"
-- Boot ab51601ddc384c01aeb840efe0b20de0 --
Dec 28 02:37:58 resin systemd[1]: Started mariner3d Printer Controller.
Dec 28 02:37:59 resin mariner[397]: Could not find platform independent libraries <prefix>
Dec 28 02:37:59 resin mariner[397]: Could not find platform dependent libraries <exec_prefix>
Dec 28 02:37:59 resin mariner[397]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Dec 28 02:38:00 resin mariner[397]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Dec 28 02:38:00 resin mariner[397]: ModuleNotFoundError: No module named 'encodings'
Dec 28 02:38:00 resin mariner[397]: Current thread 0xb6f4ff40 (most recent call first):
Dec 28 02:38:00 resin systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Dec 28 02:38:00 resin systemd[1]: mariner3d.service: Failed with result 'signal'.
Dec 28 03:04:07 resin systemd[1]: Started mariner3d Printer Controller.
Dec 28 03:04:07 resin mariner[804]: Could not find platform independent libraries <prefix>
Dec 28 03:04:07 resin mariner[804]: Could not find platform dependent libraries <exec_prefix>
"

KTheMan, 2021-12-31:
Looks like the root issue is Python, so there's a couple things you can try. I would try to completely remove and reinstall Python and its symlinks. It may also work to just reinstall the OS from a fresh download and hope that the symlink issue isn't baked into the RPiOS release.

For the former option, this is where I'd start: https://askubuntu.com/a/1202976

antwal, 2022-01-06:
same problem, try fix it with python3.9 but i receive this error:

```
mariner3d.service - mariner3d Printer Controller
     Loaded: loaded (/lib/systemd/system/mariner3d.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2022-01-06 19:44:03 GMT; 4s ago
    Process: 1273 ExecStart=/usr/bin/mariner (code=exited, status=1/FAILURE)
   Main PID: 1273 (code=exited, status=1/FAILURE)
        CPU: 425ms

Jan 06 19:44:03 raspberrypi systemd[1]: Started mariner3d Printer Controller.
Jan 06 19:44:03 raspberrypi mariner[1273]: Traceback (most recent call last):
Jan 06 19:44:03 raspberrypi mariner[1273]:   File "/usr/bin/mariner", line 5, in <module>
Jan 06 19:44:03 raspberrypi mariner[1273]:     from mariner.server import main
Jan 06 19:44:03 raspberrypi mariner[1273]: ModuleNotFoundError: No module named 'mariner'
Jan 06 19:44:03 raspberrypi systemd[1]: mariner3d.service: Main process exited, code=exited, status=1/FAILURE
Jan 06 19:44:03 raspberrypi systemd[1]: mariner3d.service: Failed with result 'exit-code'.
```

antwal, 2022-01-07:
Problem solved, 

I installed python 3.7.4 but it was very long as a process on RPi Zero W, about 4 hours to compile it complete with optimization; without optimization about ~1 hours; I highly recommend compiling it with optimization.

Use root account, build and install python then install mariner (uninstall if installed):
```
apt-get update -y
apt-get install build-essential tk-dev libncurses5-dev libncursesw5-dev libreadline6-dev libdb5.3-dev libgdbm-dev libsqlite3-dev libssl-dev libbz2-dev libexpat1-dev liblzma-dev zlib1g-dev libffi-dev libgdbm-dev libnss3-dev libreadline-dev -y
cd /usr/src
wget https://www.python.org/ftp/python/3.7.4/Python-3.7.4.tar.xz
tar xf Python-3.7.4.tar.xz
cd Python-3.7.4
./configure --enable-optimizations
make -j $(grep -c processor /proc/cpuinfo)
make altinstall
ln -s /usr/local/bin/python3.7 /usr/bin/python3.7
ln -s /usr/local/bin/pip3.7 /usr/bin/pip3.7
ln -s /usr/local/lib/python3.7/ /usr/lib/python3.7
```

Optionally: Delete the source code and uninstall the previously installed packages

```
cd /usr/src
rm -r Python-3.7.4
rm Python-3.7.4.tar.xz
apt-get --purge remove build-essential tk-dev libncurses5-dev libncursesw5-dev libreadline6-dev libdb5.3-dev libgdbm-dev libsqlite3-dev libssl-dev libbz2-dev libexpat1-dev liblzma-dev zlib1g-dev libffi-dev libgdbm-dev libnss3-dev libreadline-dev -y
apt-get autoremove -y
apt-get clean
```

Process Status:
root@raspberrypi:~# systemctl status mariner3d
```
● mariner3d.service - mariner3d Printer Controller
     Loaded: loaded (/lib/systemd/system/mariner3d.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-01-07 22:03:54 GMT; 9s ago
   Main PID: 26272 (mariner)
      Tasks: 6 (limit: 415)
        CPU: 7.760s
     CGroup: /system.slice/mariner3d.service
             └─26272 /opt/venvs/mariner3d/bin/python /usr/bin/mariner

Jan 07 22:03:54 raspberrypi systemd[1]: Started mariner3d Printer Controller.
Jan 07 22:04:02 raspberrypi mariner[26272]: INFO:waitress:Serving on http://0.0.0.0:5000
```

Rhastlyn, 2022-07-29:
I also ran into this issue, looks like nothing has changed.  I'm looking forward to trying Mariner out -- is @antwal's fix the preferred path?

Rhastlyn, 2022-08-03:
Following antwal's process had the same end result; I can confirm it definitely takes an afternoon to execute it and I still don't have a working mariner build.  The error after following the process is the same as I got before.  I stated at that point and uninstalled Mariner, followed the process, and then reinstalled mariner.  Thanks if anyone's looking at this!

Rhastlyn, 2022-08-06:
I'm going through the process again tonight and will document and post tomorrow.  Right now it's on test 131/416 and it's been doing the tests for 3 hours.  8-)

Rhastlyn, 2022-08-07:
Ok, well I've made some progress. I'm going to keep taking a look at it and may open another issue.  Going through again after formatting the SD card seems to have helped.  Here's the steps I followed:

1. Hardware: new Sandisk 32GB micro SD Card, Raspberry Pi Zero W
2. Format Fat32, default allocation size
3. Write Raspberry Pi OS Lite 32b(20220404) to SD card with Raspberry Pi Imager v1.7.2, enable SSH, set up wifi, hostname, and pw 
4. Eject disk and install in Pi
5. Wait until green light is consistently on
6. SSH into pi at <hostname>.local with pi user
7. `sudo -s`
8. Execute 'antwal procedure' above
9. Follow Mariner installation procedure from documentation
10. Reboot
11. Confirm Mariner is running with `sudo systemctl status mariner3d`
12. Connect to pi via web browser at [http:](http://<pi ip address>:5000/)

These results are encouraging and far better than my prior attempts.  I attribute this to starting over from scratch rather than trying to bandaid something that wasn't working.

Mariner loads into web browser, however I'm not fully working yet.  Currently getting the following (wondering what that "elegoo_mars.get_selected_file" is all about.  I'm using an Anycubic Photon Mono 4k.

```
Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 46, in print_status
    selected_file = elegoo_mars.get_selected_file()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 102, in get_selected_file
    self._extract_response_with_regex("ok '([^']+)'\r\n", data).group(1)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 39, in _extract_response_with_regex
    raise UnexpectedPrinterResponse(data)
mariner.exceptions.UnexpectedPrinterResponse
```

antwal, 2022-08-07:
@Rhastlyn it is not what needed anyway if it can be of help to you I publish the information about my installation

```
root@mariner3d:~# cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

root@mariner3d:~# uname -a
Linux mariner3d 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux

root@mariner3d:~# python
python             python3            python3.7          python3.7m         python3.7m-config  python3.9          
root@mariner3d:~# python3.7 -V
Python 3.7.4
root@mariner3d:~# python -V
Python 3.9.2
```

just to be sure, have you configured the [serial port](https://mariner.readthedocs.io/en/latest/software-setup.html#setting-up-the-serial-port) correctly?

however I have not updated the operating system to the latest version because the one I used was working correctly, if I have time in the next week I download the version you used and try to see if it works

antwal, 2022-08-09:
@Rhastlyn I made a new installation and it works perfectly, the error you receive is almost certainly a serial port configuration problem or you have connected the cables wrong and there is no correct communication with the printer, check the installation and configuration procedure and definitely solve the problem.

[image]

[image]

jbohanon, 2023-11-10:
This won't be a viable solution for everyone, but after lots of headache around trying to compile and install python 3.7, I just reflashed my SD card to a fresh version of Raspberry Pi OS based on Debian Buster (the latest of which was released in 2021 and can be found [here](https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/)). On that version, the install "just works"-ish. Additional hoops I had to jump through to get the GPG key are documented [here](https://github.com/luizribeiro/mariner/issues/562#issuecomment-1804208200). 

This comes with some security risks since that release is not likely receiving security patches anymore, and we're using an expired gpg key. I decided to take these risks since it works and I'll not be connected to the internet.

amd989, 2026-11-11:
Closing: this was about installing on Raspbian buster with Python 3.7 from the old PPA. [Mariner 1.0.0](https://github.com/mariner3d/mariner/releases/tag/v1.0.0) ships for Raspberry Pi OS bookworm and trixie with its own Python, so these steps no longer apply. Install instructions: https://mariner.readthedocs.io/en/latest/software-setup.html

If this still happens on 1.0.0, comment here and I'll reopen.

closed 2026-11-11 by amd989