luizribeiro/mariner #530: Support for Mars 3

koshisan, 2022-02-11:
Hi,

I received my Mars 3 a couple of days ago and the first thing I did was to look inside to check the mainboard.

I made some pictures, but realized later that there was a cleaner one already on the Chitu website: https://www.chitusystems.com/product/chitu-m20-motherboard-use-with-6-6-4k-mono-lcd-monochrome-with-chitusystems-for-elegoo-mars3/

The layout is a bit different, but RX, TX, GND and 5V are already soldered in and labled. 

Has this printer already been tested and found not working, or am I the first one?

koshisan, 2022-02-11:
Update: Getting the hardware connected is indeed no problem, this only took a couple of minutes.

Getting the installation to work with python 3.9 (and figuring out what was wrong in the first place) took way longer. See https://github.com/luizribeiro/mariner/issues/481

Also, since the Mars 3 requires encrypted files, I ran into

https://github.com/luizribeiro/mariner/issues/180

next.

For the moment I have no problems with starting the printjob in the actual printer itself - I have to go there to deal with the resin, etc. anyway. But it would be nice to make uploading and actually starting the print two different things, so at least the upload would work without error....

jpreculeau, 2022-03-11:
Exactly the same for the new Saturn S. I guess if the Mars 3 get supported then the Saturn S will be too.

wizard311, 2022-05-08:
I own the Saturn S and I know it reads CTB v4 and it's encrypted. mariner3d does not play well with it regardless of if chitubox or lychee created the file.  I read somewhere on here that mariner3d cannot parse these files yet.  I know a branch was working on talking to chitu boards with firmware 5.4 and the Saturn S uses firmware 5.5 so I thought that might be an issue as these firmware use the encrypted CTB v4.

JimKnopfIoT, 2022-06-05:
I do my slicing with Prusaslicer (creates an .sl1 file) and use UVtools to convert it to encrypted .ctb files. UVtools may have some useful code for handling encrypted .ctb files.

jremen, 2022-08-17:
@JimKnopfIoT Probably only using closed SDK from Chitu. Which for many in OSS is no no due to politics, nothing else.
I would like to see these OSS projects would just download SDK remotely upon user aggreement during installation and we would be fine I guess. 

For now, it seems both Mariner and Octoprint Chitubox plugin are dead. Developers not responding, nobody doing any branches…