luizribeiro/mariner #543: Error 500 + python error

jpreculeau, 2022-03-07:
**Description**
Hello,

I'm on a raspberry PI zéro W2 with the rasbian legacy lite installed.
I've ma,age to install mariner + samba for my Saturn S.
My main problem 👍 
When loading my Ip:5000 it's working unless I try to go tot the files section or to upload a file.
Then I got an error message and an endless charging.
the msg :
 Something went wrong
The server replied with a 500 HTTP status code.

i've also lauch the command line to check if the service is working and there is a weird last line with an error. 

pi@SaturnMariner:~ $ sudo systemctl status mariner3d
● mariner3d.service - mariner3d Printer Controller
   Loaded: loaded (/lib/systemd/system/mariner3d.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2022-03-05 03:05:38 CET; 2 days ago
 Main PID: 332 (mariner)
    Tasks: 6 (limit: 871)
   CGroup: /system.slice/mariner3d.service
           └─332 /opt/venvs/mariner3d/bin/python /usr/bin/mariner

Mar 07 20:27:22 SaturnMariner mariner[332]:     FILES_DIRECTORY / selected_file
Mar 07 20:27:22 SaturnMariner mariner[332]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask_caching/__init__.py", line 952, in decorated_function
Mar 07 20:27:22 SaturnMariner mariner[332]:     rv = f(*args, **kwargs)
Mar 07 20:27:22 SaturnMariner mariner[332]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/utils.py", line 20, in read_cached_sliced_model_file
Mar 07 20:27:22 SaturnMariner mariner[332]:     return file_format.read(FILES_DIRECTORY / filename)
Mar 07 20:27:22 SaturnMariner mariner[332]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/file_formats/ctb.py", line 135, in read
Mar 07 20:27:22 SaturnMariner mariner[332]:     ctb_slicer = CTBSlicer.unpack(file.read(CTBSlicer.get_size()))
Mar 07 20:27:22 SaturnMariner mariner[332]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/typedstruct.py", line 89, in unpack
Mar 07 20:27:22 SaturnMariner mariner[332]:     return cls(*unpack(cls.get_format(), buffer))
Mar 07 20:27:22 SaturnMariner mariner[332]: struct.error: unpack requires a buffer of 76 bytes
pi@SaturnMariner:~ $

In the positive side : I can print files form the Pi without problem when using the printer's screen.

wizard311, 2022-04-24:
I have your exact setup but Mariner web interface is not working for me
I can load prints to pi via smba network and manually start the print from the Saturn S.
Running mariner3d_0.2.0-1_armhf.deb

When I start the web interface I get 

Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 46, in print_status
    selected_file = elegoo_mars.get_selected_file()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 102, in get_selected_file
    self._extract_response_with_regex("ok '([^']+)'\r\n", data).group(1)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 39, in _extract_response_with_regex
    raise UnexpectedPrinterResponse(data)
mariner.exceptions.UnexpectedPrinterResponse

and if I go to the file section of web interface I get

The server replied with a 500 HTTP status code.

Not trying to hijack your thread I just could not figure out how to start my own and I saw you have similar things going on.  How did you get the web interface to load without errors?

wizard311, 2022-04-25:
To follow up, I have this as well "Mar 07 20:27:22 SaturnMariner mariner[332]: struct.error: unpack requires a buffer of 76 bytes"  I am not a programmer or coder but from what little research I could look up on the web.  It appears Mariner is not completely loading due to lack of access to a Root Buffer.  If I am correct this is the cause of both of our issues.

JimKnopfIoT, 2022-06-19:
Same here on Mars3. I compiled python 3.7.4 on a RPi4 and copied the files to the Pi Zero 2 W to speed up things. I'm able to start mariner3d. When accessing the Web GUI i only can upload files. I can't see any other menu or information. After uploading i get the Error 500 message. I can't see or remove the files on the Pi Zero virtual USB drive. The files are there. I can delete them by hand in /mnt/usb_share via ssh or WinSCP. The GUI is broken.

jonboy86, 2023-08-26:
same here wonder if anyone has sorted it

amd989, 2026-11-11:
Fixed in [Mariner 1.0.0](https://github.com/mariner3d/mariner/releases/tag/v1.0.0): the file list no longer fails when one file can't be read. A file Mariner can't parse is listed as not printable instead of the whole page returning a 500, and encrypted `.ctb` files from newer ChiTuBox and Lychee versions are now supported.

If this still happens on 1.0.0, comment here with the log and I'll reopen.

closed 2026-11-11 by amd989