luizribeiro/mariner #586: After install following guide, mariner process exits on boot

idesignstuff, 2025-03-30:
**Description**

pi@Saturn:~ $ sudo systemctl status mariner3d
× mariner3d.service - mariner3d Printer Controller
     Loaded: loaded (/lib/systemd/system/mariner3d.service; enabled; preset: enabled)
     Active: failed (Result: signal) since Sat 2025-03-29 22:59:22 CDT; 1min 25s ago
   Duration: 546ms
    Process: 363 ExecStart=/usr/bin/mariner (code=killed, signal=ABRT)
   Main PID: 363 (code=killed, signal=ABRT)
        CPU: 47ms

Mar 29 22:59:22 Saturn systemd[1]: Started mariner3d.service - mariner3d Printer Controller.
Mar 29 22:59:22 Saturn mariner[363]: Could not find platform independent libraries <prefix>
Mar 29 22:59:22 Saturn mariner[363]: Could not find platform dependent libraries <exec_prefix>
Mar 29 22:59:22 Saturn mariner[363]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Mar 29 22:59:22 Saturn mariner[363]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Mar 29 22:59:22 Saturn mariner[363]: ModuleNotFoundError: No module named 'encodings'
Mar 29 22:59:22 Saturn mariner[363]: Current thread 0x76f4f040 (most recent call first):
Mar 29 22:59:22 Saturn systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Mar 29 22:59:22 Saturn systemd[1]: mariner3d.service: Failed with result 'signal'.

**Reproduction Steps**
Steps to reproduce the behavior:
1. Follow install instructions
2. reboot
3. run sudo systemctl status mariner3d
4. See error above

**Expected Behavior**
I expected mariner3d to run, but it doesn't
**Screenshots / Logs**
If applicable, attach screenshots or logs to help explain your problem.

**Details:**
 - Mariner Version: 0.2.0-1
 - Printer Model: Elegoo Saturn v1
 - Raspberry Pi Model: RPi Zero 2W with fresh raspi os 32 bit lite image from rpi imager

**Additional context**
Add any other context about the problem here.

DanyloKh, 2025-05-20:
Do you fixed the problem? I have exactly the same...

chicco83, 2025-12-15:
same problem for me

amd989, 2026-11-11:
Closing: this was about installing on Raspbian buster with Python 3.7 from the old PPA. [Mariner 1.0.0](https://github.com/mariner3d/mariner/releases/tag/v1.0.0) ships for Raspberry Pi OS bookworm and trixie with its own Python, so these steps no longer apply. Install instructions: https://mariner.readthedocs.io/en/latest/software-setup.html

If this still happens on 1.0.0, comment here and I'll reopen.

closed 2026-11-11 by amd989